mikemag/MonkeyCAM #23: Difficulty Getting Started can not get the program to run.

mikemigs10, 2015-01-12:
Back Again and I am still having difficulty.  I will post the same problem on Github shortly after this post.  If someone could tell me what I am doing wrong that would be great.  Because right now I have a cmc mogul that is in need of some use.

So I read through the thread and the documentation on github and believe I have followed the direction (obviously not).  After downloading the .dmg file I then took the folder and put it into a folder I created on my desktop called MonkeyCam.  After that I opened the terminal and pasted the code that github provided.  

cd <directory where you copied the dmg contents>
mkdir SampleBoard
bin/MonkeyCAM --board example/board-def.json --machine example/machine-tool-def.json  --outdir SampleBoard
open Sample-output/Sample-snowboard-overview.html

Changing <directory where you copied the dmg contents> to 

cd <MonkeyCam>
mkdir SampleBoard
bin/MonkeyCAM --board example/board-def.json --machine example/machine-tool-def.json  --outdir SampleBoard
open Sample-output/Sample-snowboard-overview.html

this did not work. I then took all the files out of their folders so they were just sitting in the MonkeyCam folder. and posted the code again and it did not work. When I opened the terming all I did was post the code. Are there other inputs that I need to put in before I post the code?

What am I doing wrong?

thanks for the help!

splitn2, 2015-01-12:
what errors did you get when you executed the commands?
can you cut and past the exact contents from your terminal session please.
cheers
Rich

mikemigs10, 2015-01-12:
this is what is posted after I paste and press enter

mike-migs:~ mikemigliorino$ cd <MonkeyCam>
-bash: syntax error near unexpected token `newline'
mike-migs:~ mikemigliorino$ mkdir SampleBoard
mkdir: SampleBoard: File exists
mike-migs:~ mikemigliorino$ bin/MonkeyCAM --board example/board-def.json --machine example/machine-tool-def.json  --outdir SampleBoard
-bash: bin/MonkeyCAM: No such file or directory
mike-migs:~ mikemigliorino$ open Sample-output/Sample-snowboard-overview.html

splitn2, 2015-01-12:
I think you are cut and pasting the commands, which is causing issues here.

instead I recommend you run one command at a time

don't run the cd command its unnecesary
mike-migs:~ mikemigliorino$ cd 
-bash: syntax error near unexpected token `newline'

its telling you the SampleBoard directory already exists so no need to run that command
mike-migs:~ mikemigliorino$ mkdir SampleBoard
mkdir: SampleBoard: File exists

Something is not right here its telling you it cannot find the MonkeyCAM executable I think, so check your files are in the right place. Personally I found during testing easier to lump the MonkeyCAM executable file in with the board and machine defs files and run all out of the same directory then no need to put the extra path info in
mike-migs:~ mikemigliorino$ bin/MonkeyCAM --board example/board-def.json --machine example/machine-tool-def.json --outdir SampleBoard
-bash: bin/MonkeyCAM: No such file or directory
mike-migs:~ mikemigliorino$ open Sample-output/Sample-snowboard-overview.html

worst case we try set up a remote terminal session so I can help you debug this issue online???

cheers
Rich

mikemigs10, 2015-01-12:
This is going to take me a min to process. I thought that was all one command not multiple commands.  I will give this a shot.  One thing though, when you say MonkeyCam Executable file do you mean the file that you would select to run the actual program?  Ill get back soon.

splitn2, 2015-01-12:
haha all good Mike I understand! yes MonkeyCAM exutable is the actual binary program and the other two "defs" files are your board and machine parameters. Once you get it right once you'll be sorted. So envious you have your cnc ready to go mine is still a wee way off, I think over engineering it is actually slowing me down.......

splitn2, 2015-01-13:
Mike I think we could try screen sharing with Skype if you need to, in the next day or 2 but not tonight sorry I am busy. cheers

splitn2, 2015-01-13:
@mikemigs10 are you winning? real keen to see how you get on with your new cnc machine! Are you running it with MACH3 or MACH4 or DESKCNC or something else? cheers Rich

mikemigs10, 2015-01-13:
I did not win last night or this morning but I did make some progress mentally, I think.  I was playing around with the codes that you broke down for me and trying to figure out why they worked and why they didn't work.  If you would be willing to setup a Skype session with me and run me thru everything I would be so grateful.  I can make myself available anytime this week so whatever your schedule is like just let me know. Unfortunately tonight does not work for me.   I live in Connecticut USA if there is a timezone difference.    I am running Mach3 for my cnc.  I have been using it but for very basic things like cutting shapes.  I have not figured out how to trim the materials yet, from the sounds of it this program will take care of that learning curve for me.    Rich thanks for the help!

splitn2, 2015-01-14:
@mikemigs10 I think I am 18 hours ahead of you so mid afternoon your time would be later evening my time. Can you please email me your skype id to sales@splitn2.com and I'll try look you up on skype ok, Then we can try later this week and see how things go. cheers Rich

splitn2, 2015-01-14:
@mikemigs10 I have got Skype on my pc and I think I have sent you a contact request there were 2 guys with your name...... cheers Rich

mikemigs10, 2015-01-14:
I was having problems with my Skype account and the sign in so I created a new one my name is mike.migliorino36.   I am around right now  till about 230 my time or if tomorrow works for you I got all day.

splitn2, 2015-01-14:
@mikemigs10 haha I did some googling and found Facet Skis!!!!! nice work mate love to see the entrepreneurial spirit hard at work. I'll try hook up on your skype tonight ok and then lets try meet from there. cheers Rich

splitn2, 2015-01-18:
@mikemigs10 I have set up a ".bash_profile" file for my username on my mac which appends the path to find MonkeyCAM executable to the default paths, this means I can execute MonkeyCAM from any directory without prefixing it with a path statement and it will run. All I need is to reference to the paths for the board and machine defs files. I'll help you with this next time we catch up on skype ok. Cheers Rich

mikemag, 2015-01-25:
I understand Mike & Rich hooked up over Skype and got this resolve. Mike, let us know if you run into any more issues!

closed 2015-01-25 by mikemag

mikemigs10, 2015-01-26:
Hi Mike,

Thank you for reaching out. Yes, Rich helped me with the issue.  This is
amazing, big learning curve for me but awesome.

thanks again,

Mike Migliorino