mikemag/MonkeyCAM #25: Enhancement Request - CAMAX style insert holes

splitn2, 2015-01-30:
I use CAMAX inserts and I'm sure a few others do too. I'll measure up a sample and put the dimensions here. It would be great to have the flange cut set up for a non-dished flange , if that makes sense!? cheers, Rich

mikemag, 2015-01-31:
Great idea. Post up a dimension, and a couple of pics of them from different angles so I'm sure what I'm dealing with. Should be an easy addition.

splitn2, 2015-02-01:
haha love it when you say easy Mike, easy if you know how !!!! what you are doing is so clever it blows my mind, I am so looking forward to using this g-code for my board profiling and cutting and drilling!

splitn2, 2015-02-02:
Insert dimensions
 
flat to flat = 16.8mm
main diameter of flange = 19.4mm
inner barrel diameter = 9.85mm

[image]
[image]
[image]

RegMeasures, 2017-07-20:
I've just started having a look at the code which creates the shape of the insert hole (mostly in [shapes.cpp](https://github.com/mikemag/MonkeyCAM/blob/master/core/shapes.cpp)) to think about how the shape could be made more user definable. This could address this issue, as well as allowing other insert shapes as well... My priority is to make inserts with a flat, rather than bowl shaped flange. 

At the moment MonkeyCam allows input of `insert rim depth`, `insert rim diameter` and `insert hole diameter` which is great. However, MonkeyCam has a hard coded curved 'bowl' as described [here](http://www.happymonkeysnowboards.com/wiki5/index.php?title=CNC_Core_Cutting#Inserts_and_alignment_holes).

It shouldn't be too hard to change this but it will need more inputs somewhere...? One option could be to add a `insert-shape` keyword where you could choose `"bowl"` or `"flat"`? Also we could maybe move the `insert rim depth`, `insert rim diameter` and `insert hole diameter` settings into the binding-def.json file rather than the machine-def.json as they relate to the binding, rather than the machine. I.e. you may want to use different inserts for ski and snowboard bindings but make both skis and snowboards on the same machine? 

Thoughts anyone?

mikemag, 2017-07-20:
There are so few kinds of inserts I think it'd be fine to actually add an `insert shape` parameter like you suggest, and have a valid value be `CAMAX` ;) Then you could get the flat bottom of the hole, and the flat sides.

On moving those parameters, that'd be fine. I often thought I'd make a `materials-def.json` file where you'd describe all the materials you use, like base material width and thickness, same for your core blank, insert dimensions, and even fiberglass width and required margins, etc. but I just never got around to it and left the few I was using in the machine def. The idea was not only would it describe the parameters of materials required for proper machining (the only ones that exist in the machine def now), but also the parameters we'd need to validate that your materials are in fact big enough to support the given board design. That way it would shout at you if you were about to use a core blank that was too thin, or start on a core for a board that you don't have base material or glass to actually build correctly. 

Obviously adding a full-on materials def is much more work than your suggestion. Feel free to leave the parameters where they are, or move them to the binding def, or to go nuts on a materials def addition.

splitn2, 2017-07-21:
sounds great !!! around the "insert rim diameter" lets leave some room for glass to wrap from the core bottom up around and over the insert rim and flange  , same as we do for the insert rebate, a few mm wider 2-3mm I suggest. awesome work guys!