mikemag/MonkeyCAM #27: error when running monkeycam

sou-l, 2015-05-14:
Having an issue getting the program to run. I am getting the following error when trying to execute.

tried rich's fix of running in the same directory to no avail.

Using board 'board1.json', machine 'machine1.json'
libc++abi.dylib: terminating with uncaught exception of type boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::property_tree::json_parser::json_parser_error> >: machine1.json: cannot open file
Abort trap: 6

all help appreciated.
Cheers
Liam

splitn2, 2015-05-14:
hey @sou-l  mate I also had an error similar when some of my measurements were a little bit out of bound and not compatible, so you might be using the wrong measurement units
some are in mm , eg sidecut radius
some in cm , eg tip length , tail length
taper is in inches!!!!
set all your handles to 0.5 for a start and tweak from there
this should work
If in doubt try running MonkeyCam with the original board-defs.json file
Otherwise it may be a file path issue but I think Mike sorted that out
I'dd do a screen share with you from home and show you mine working I was fiddlign with it last night showing a guy

cheers
Rich

splitn2, 2015-05-14:
@sou-l  possibly a syntax in your command line or maybe a typo in the actual machine-defs.json file, it seems to be failing parsing the machine-defs.json file anyway "machine1.json: cannot open file" ?????

splitn2, 2015-05-14:
hey @sou-l make sure you "watch" and "star" the MonkeyCAM and SnoCAD-X repositories too it all helps to get the visibility up for you to see changes and also for other followers of yours on GitHub to see what you are watching, in theory that is!

sou-l, 2015-05-14:
cheers Rich. Will have another go at lunch time and see how I get on. Didn't realise taper was in inches! That's one to remember .

Will go back to the original files and make sure all is ok. Will let you know how I get on.

splitn2, 2015-05-14:
haha classic @sou-l yeah I know @mikemag has a long list of things to tweak and change he's a busy man so we need to get more experienced coders in to help on the project and let Mike keep an overall view of where its going so he can guide and steer the project and we get more coding enthusiasts who like board building to do the donkeywork!!!

sou-l, 2015-05-15:
all fixed. cheers

closed 2015-05-15 by sou-l