mikemag/MonkeyCAM #32: Tool names in machine-tools.defs

splitn2, 2016-03-28:
I will add to this issue for now its just a place holder while I am thinking of it. I was trying to enter new names and dimensions for my metric cutters and MonkeyCAM was hating on the names I was using, I will try again and add detail to this issue......

Its probably finger trouble but I thought my editing skills were better than that, do I need to use "vi" editor I think I can still remember how.......haha!!!

cheers
Rich

Tools

The tools section has an entry for each tool which will be used by the generated G-Code programs. Each tool is given an id which is used to reference the tool from other parameters, and is called the tool number in the documentation for those parameters.

"tools":
[
    {
        "id": 1,
        "name": "Quarter Inch Upcut Spiral",
        "diameter": 0.247,
        "G-code #": 1
    },
    {
        "id": 2,
        "name": "Quarter Inch Straight",
        "diameter": 0.250,
        "G-code #": 1
    },

]

mikemag, 2017-08-25:
@splitn2 Hey man, do you have any more detail on this. I know it's old, but if there's something I can do to make this easier lemme know.

splitn2, 2017-08-27:
must have been finger trouble @mikemag , last time I edited machine defs it worked fine....... , issue can be closed..... cheers

closed 2017-08-27 by mikemag