mikemag/MonkeyCAM #36: Understanding Overall Shape Dimensions

garethbradley, 2017-03-23:
Hi,

I'm working on an educational project to port MonkeyCAM into a serverless web app - I say port, but really it's using the excellent work on MonkeyCAM as inspiration, but building from scratch.

Whilst checking the output of my boards / skis to MonkeyCAM's, there's a discrepancy in the nose width & tail width.

Here's an example:
```javascript
{
    "board": {
        "name": "Kids Skis",
        "nose length": "14",
        "effective edge length": "104",
        "tail length": "6",
        "sidecut radius": "1250",
        "waist width": "8",
        "taper": "1",
        "nose shape": {
            "type": "Basic Bezier",
            "end handle": "0.75",
            "transition handle": "0.5"
        },
        "edge shape": {
            "type": "Basic Arc"
        },
        "tail shape": {
            "type": "Flat",
            "flat width": "5.5",
            "end handle": "0.50",
            "transition handle": "0.75"
        },
        "nose and tail spacer width": "2",
        "tail edge extension": "2"
    },
    "profile": {
        "nose thickness": "0.3",
        "center thickness": "0.79",
        "tail thickness": "0.3",
        "nose taper": {
            "taper start": "0.45",
            "start handle": "0.70",
            "end handle": "0.80",
            "taper end": "1.00"
        },
        "tail taper": {
            "taper start": "0.51",
            "start handle": "0.68",
            "end handle": "0.85",
            "taper end": "1.00"
        }
    }
}
```

Using MonkeyCAM, this produces the following summary:

- Total length: 124.0000cm
- Nose / effective edge / tail length: 14.0000cm / 104.0000cm / 6.0000cm
- Nose / waist / tail width: 10.6642cm / 8.0000cm / 9.6642cm
- Taper: 1.0000cm
- Sidecut radius / depth: 1250.0000cm / 1.0821cm
- Reference stance width: None
- Setback: None
- Board area: 1082.687cm2
- Extension of metal edge towards nose: None
- Extension of metal edge towards tail: 2.0000cm

Using my webapp, I get the following:
- Total length: 124.0000cm
- Nose / effective edge / tail length: 14.0000cm / 104.0000cm / 6.0000cm
- Nose / waist / tail width: 10.16cm / 8.0000cm / 9.16cm
- Taper: 1.0000cm

When I draw it up in SketchUp, I've measured the expected Nose / Tail widths to be 10.26cm / 9.26cm.

So, any idea what the discrepancies are? I'm pulling my hair out trying to find the bug...

Many thanks,
Gareth

splitn2, 2017-03-28:
@mikemag this looks interesting!!!!! @garethbradley welcome and it great to see you working on MonkeyCAM, whats your interests and background? cheers, Richard

garethbradley, 2017-03-29:
Hi Richard,

Thanks for the warm welcome! In terms of interests and background - I've got a fascination with serverless architectures. I run online services for a FTSE 100 and so a large part of what I do is around security, patching, performance optimisation and automation of services. I wanted to find a real world example where serverless functions (in this case, AWS Lambda) could be used to create an entire web app.

I've started with an extendable set of "processors" which are the tail types (Bezier), nose types (Bezier, Flat) and sidewall patterns (Arc). The processors are called from various image generators (board overview, top profile, profile, core etc.) All of this goes though an API gateway, so the idea is that there is, essentially, an online & scalable API for ski design / CAM.

Future plans would be a one stop online design studio for designing boards / skis and visualising them (3D maybe?), then producing the necessary CAM. I'm a LONG ways off that though!

But, basically, I'm just into tinkering with things. Building skis is the latest obsession to join skiing, mountain biking, SCUBA, building HiFi / Audiophile equipment, electronics & programming! 😆

splitn2, 2017-03-29:
wow @garethbradley you are talking way above my pay grade mate, welcome welcome welcome!!!! Hey you need to talk to @mikemag and get rolling on some of this while you are super enthused!! I am not a coder merely a wrangler of clever guys like you..... Ideally we keep a common set of goals and efforts and I suspect between you both perhaps some discussion for starters on your architecture and ideas so we can as much as possible make all time spent by you and others all in a good common direction..... hows that ? In case you've not seen it check this too.... https://github.com/mikemag/MonkeyCAM/wiki/Design-Overview cheers, Rich

RegMeasures, 2017-06-12:
Hi @garethbradley, 

I'm just getting back into doing a bit more developing on MonkeyCam and thought I'd have a quick look at this issue. Although you may have resolved it already I guess? 

Anyhow, after a bit of digging I think there must be an error in the logic you use to calculate the curvature or taper of the effective edge. I while ago I made a simple spreadsheet to calculates radius of curvature from tip, tail and waist widths. If I input MonkeyCam's dimensions and your webapp dimensions the MonkeyCam radius is 125cm but your webapp radius is 163cm, larger than in the json file. Looking at your sketchup drawing shows a tip and tail width of 101.6mm if there is no taper. As a mental exercise we could approximately adjust for 10mm taper by rotating the edge about its centre so that the tip gets wider and the tail narrower. Given that we are rotating around the centre, the tip is going to get wider the same amount the tail gets narrower i.e we need to make the tip 5mm wider and tail 5mm narrower to achieve a 10mm difference (taper). Hence Tip = 106.6mm, Tail = 96.6mm. Almost exactly what MonkeyCam calculates.

@mikemag, correct me if I am wrong but I think one point worth noting is that in MonkeyCam the "waist width" is actually the width of the centre of the effective edge. Allowing for taper, the narrowest point of the ski is usually slightly behind the centre of the effective edge, and hence slightly narrower than the "waist width" entered into the *.json file. I think the difference is generally pretty small unless you have a really extreme taper. This simplification makes MonkeyCam more reliable as it allows it to deal with situations where the user specifies an extreme taper (e.g. such that the narrowest part of the ski/board the tail).

Hope that all makes sense.

garethbradley, 2017-06-12:
Ahhh! Thanks for that :)

I was simply subtracting the taper from the rear, rather than effectively rotating the side wall about it's centre. Harder to calculate it, but I'll jump back in soon and see what I can do to resolve it.

Thanks for the time & explanation!

mikemag, 2017-06-12:
Sorry for the delay on this one. @RegMeasures is 100% correct in his explanation. His description of "rotating the edge about its centre so that the tip gets wider and the tail narrower" is an easy way to think about how to add taper to a board.

closed 2017-06-12 by mikemag