mikemag/MonkeyCAM #37: Binding inserts for skis...

RegMeasures, 2017-06-12:
Most ski's don't have inserts as standard, but a few do. I figure ski design has learnt a lot from snowboards, so why not... Been discussing this with @splitn2 for lightweight cores which would otherwise need a lot of reinforcement to prevent pull out.

The obvious challenge for implementing is that ski bindings are generally designed for a specific and exact hole pattern so a generic pattern of inserts won't work. This means that skis need to be custom built for a specific binding (and potentially boot length depending on how adjustable the binding is). But this also makes it ideally suited for CNC as the inserts can be precisely positioned.

After spending a bit of time looking at the way the snowboard inserts are set up in MonkeyCam I was thinking:

- "stance setback" could be used to define the setback to the ski boot centre from the effective edge centre.
- Rather than messing up the existing "nose insert pack" or "tail insert pack" I should create a new "ski insert pack" section in the board def json file.
- The "ski insert pack" will need a setting for "boot length".
- There needs to be some way to define the hole layout manually or from a library - this might be a bit complex to fit into the board def json file so maybe I should put this in a seperate binding def json file which gives details of the screw pattern, that way implementing a new binding would just require creating a new binding def file and then refer to it in the board def file? What do you think @mikemag. To specify the hole locations I'm going to have to define a list of x-y positions relative to boot toe and heel centres I think.

Any thoughts @mikemag? I guess we could leave this mod in a standalone fork rather than merge it into the main code as it is a bit specialist but i figure I should try and do a tidy job of it as if it is going to be imported? Where do you think the best place to input the hole pattern for a specific binding is?

RegMeasures, 2017-06-12:
Just realised I forgot to mention that @splitn2 has started pulling together a database of binding hole patterns here [https://github.com/splitn2/DrillSki](https://github.com/splitn2/DrillSki).

RegMeasures, 2017-06-24:
Hi, 
I've made a start on implementing this in a [new branch](https://github.com/RegMeasures/MonkeyCAM/tree/BindingInserts) and I've got MonkeyCAM accepting a new (optional) `--binding` command line input.

I have a few decisions to make where I think some advice would be useful. I'm basically a little bit lost with the object orientated programming! 

As I understand at the moment there is an object `shape` of class `BoardShape` which holds all the information relating to the board and another object `machine` of class `Machine` which holds all the information relating to the machine. The `BoardShape` class currently contains (optional) `nosePack` and `tailPack` objects of class `InsertPack`.

I need to decide where to store information from the binding-def.json inputs. First question: is my basic summary of the overall structure rightish? Second question: where do you think it would be best to work for my current changes - I'm thinking at least some of what I will be changing needs to be inside the `InsertPack` class?

Any input much appreciated. Feeling a little bit out of my depth but that's how to learn!

Reg

RegMeasures, 2017-07-02:
Ok - I've made quite a bit of progress on this over the weekend and although I'm sure I've not quite got the best way of passing variables around at the moment I am starting to see some results (See my [BindingInserts branch](https://github.com/RegMeasures/MonkeyCAM/tree/BindingInserts)). 

One question regarding input data:
Binding hole position sign convention etc.: In [DrillSki](https://github.com/splitn2/DrillSki) the system appears to be X positive towards the tip and relative to pin position on the toe for pin bindings, true toe position for non-pin bindings. Units are mm. This seems logical to me but MonkeyCam currently has sign convention X positive towards the tip and dimensions in cm, which could be a source of confusion? In my current workings I have gone with the DrillSki sign convention (positive towards tip) and units (mm) in the binding-def.json file. What do you think @mikemag @splitn2? 
The use of pin position rather than toe position for pin bindings seems totally logical and I am keen to stick with it (even though it means boots will need to be measures accurately rather than just reading off the "sole length" marked on them.

mikemag, 2017-07-04:
Hey man, sorry for not getting back to you sooner. I've looked at all of your commits into your branch, and I like the direction you've been taking. Nice work!! I love that shot you have above with the ski insert holes in place :) 

To your questions:
* The issue with the X coordinates is important. I usually find that consistency is key. The stance setback is defined as positive values offset towards the tail. Though not labeled as a X coordinate it essentially is. The axis labels on the output drawings also show X positive towards the tail. I recommend you stick with this, however...
  * [DrillSki's coordinate def](https://github.com/splitn2/DrillSki/blob/master/variables) is backwards in both axes from MonkeyCAM. They define +X as towards the nose, and +Y as towards the left of the ski. I suspect this comes from rotating the binding template 90 degrees clockwise, to overlay a ski laying in front of you with the nose to the right. I lay down boards in front of me with the nose to the left, because I ride snowboards regular footed and not "goofy" ;) 
  * Honestly, I find the DrillSki coordinate system choice odd. Given that binding templates are usually presented in PDF format, with the nose to the top of the page, I would have expected +X towards the right of the ski and +Y towards the nose of the ski. @splitn2, what's the rationale there? Is it just how you lay the board down vs. looking at the PDF?
  * I think if you want to specify the binding locations relative to a different coordinate system than what MonkeyCAM is already using, then we can do that, but we should make sure to make this clear in the binding def. Perhaps a `coordnate-system` parameter with values of `tip-left`, `tip-right`, or `tip-up`? 
* Units: please stick with cm for now, for consistency with the board/ski def. I plan to go thru sometime and make every input include the units (`"`, `in`, `cm`, or `mm`), then we never have to assume again. But that's not done yet.
* On pin position vs. toe position, I'm in no position o say! (Pun intended.) Not being a skier, I shall defer to you guys :)

Again, thanks for your work on this!!

RegMeasures, 2017-07-08:
Hey, thanks for your input, glad you like the picture :)

I agree that consistency with the coordinate systems is important. I guess whatever I do it will have to be well explained in the documentation. I'll flip things around to use cm and X positive towards the tail consistent with the rest of MonkeyCam and tidy things up. I suspect whatever I do it will be counter intuitive to somebody! I don't think I’ll bother with a coordinate system parameter in the binding def as it seems like unnecessary complexity which could just cause even more confusion.

Including units for all measurements would be AWESOME! I know @splitn2 would appreciate the option to use mm for some of the machine def and I definitely enter mm by accident for thing like waist width pretty frequently.

splitn2, 2017-07-19:
sorry @RegMeasures @mikemag I have not logged in to Github for ages so missed all the action, although Reg did tell me all this the other night. No surprise to me everything I've done in my Drillski stuff would need checking in terms of x and y direction and funnily enough when I mount skis I normally sit the ski on my bench with the tip left and tail right ....... cheers guys and thanks for the great work on this stuff. I look forward to these improvements making skis and boards in metric might be fun one day :) and rapid splitboard designs and builds too one day haha! I'm never gonna make a coder and I really appreciate the efforts going into this, I'll keep trying to recruit more coders and testers.....

mikemag, 2017-08-14:
@RegMeasures I've merged your pull request. Shortly I'll put up a new release, and I'll add the Mac executable into it. I'd like you to add in the Windows distribution if you wouldn't mind :) (It's the reason I added you as a collaborator, you'll be able to add to/edit the release.)

splitn2, 2017-08-17:
@mikemag  @RegMeasures great to see another release awesome work guys!!!! Reg we need to carry on with your project skis now, have you defined the insert dimensions too for the inserts we'll use , the stainless M5 tee-nuts ?

mikemag, 2017-08-25:
Included in v4.0.6.

closed 2017-08-25 by mikemag