mikemag/MonkeyCAM #41: Splines , sidecuts, profiles ?????

splitn2, 2018-03-06:
Hi team - @RegMeasures @mikemag @NickAlbers , I've had a lot of fun lately with Fusion360 and splines , mainly this started with helping George Gaines @ Chilton Skis with his ski press profile redesign, albeit badly I think I came up with better than he had originally. But my curiosity with splines actually goes back to Styrotech Ltd who were a NZ based highly successful windsurf custom race board company in the 80's and 90's until windsurfing died..... I used to watch Kevin Trotter the designer/owner playing with splines on MacSurf a marine design program for Mac computers when I was flatting with Kevin in the early 90's. It all went over my head I just know it was a big part of his boards being fast. So recently I was helping some young guys build some power boards for Japan and they used Fusion360 and splines to do their sidecuts and press profile camber/rocker press blocks and they came out beautifully. Then helping George I used splines to redesign his ski press profiles and then put them through the Fusion360 "Curvature Comb Analysis" tool to check for bumps and hard to see features that are in the splines - bumps that are mathematical calculated and are fair but not necessarily what you want in an efficient snow or water craft where we don;t want drag or an effect of "pushing" snow ...... Again I must say I am a total rookie at this and its total gut feel but given the boaties use I can't help but think its worth exploring.

So I am wondering to really round out the capabilities of MonkeyCAM is there any agreement with you guys in the effort of using splines in MonkeyCAM for press profiles and sidecuts

Taking that further , is there a way we can have a continuous spline from the tip all the way to the tail end to end ??? Maybe not so easy to define parametrically in MonkeyCAM currently ???

And is the code available to use hedgehog style comb curve analysis and can it be brought into MonkeyCAM???

What do you guys think? I know you are all a way smarter with your maths than me!!!!!

mikemag, 2018-03-09:
I think there are some interesting possibilities here for sure. 

Currently MonkeyCAM uses splines in a few places internally, and two obvious places which are user-controlled: the core profile and the nose/tail shapes. Using the profile as an example, this is made up of a couple of classic Bezier curves (https://en.wikipedia.org/wiki/B%C3%A9zier_curve): two end points and two control points. I constrain the control points to ensure that the splines meet properly in the middle of the board (no kinks) and to ensure that they meet the ends properly as well. By tweaking the control points you can end up with a profile resembling something close to a smooth arc, or a flat section in the center with a smooth taper down to a flat section at the ends, etc. 

The problem with the approach to both the profile and end shapes is I fear the problem we'll have using splines in more places: it's pretty imprecise. You basically tweak the numbers until the shapes "look right". I'd like to see more control where I can define what "right" is more numerically.

Also, as you'll notice, the profile and end shapes use just one Bezier each (per half). It would be great to be able to use multiple so we can get more complex shapes from the nose/tail, and profiles with thinner centers.

I'd also love to see more complex edge shapes. Are these made up of splines, or multiple arcs of well-defined radii blended together in some fashion? I often hear people talk about having an edge which is, say, 9m at the front and 8m at the tail. How does that blending occur? What control do you place on it, i.e., how do you define how quickly the blend takes place? Do you blend more than two radii? Etc. On the spline side of it, I've used a Catenary for edges before (https://en.wikipedia.org/wiki/Catenary), and a parabola. However, when I did this I formed the nose/waist/tail widths via an arc, and then essentially fit the other shapes to those points. It didn't result in what I saw as an interesting difference vs. a pure arc, so I stopped using it (and did not port it to the current version of MonkeyCAM.)

Would love to hear other ideas here.

This reminds me of something I need to do to MonkeyCAM in general, something it used to do: allow comparison between shapes. It's really helpful to visualize the difference in end or edge shapes, and to get a quantification of their difference along their lengths. I'll open an issue for that.

splitn2, 2018-03-11:
I totally agree with what is right numerically, if we get a combination of what looks right to the eye and in fact numerically is goodness, then we are in better shape surely . Taking my badly explained ideas on why splines are good and how the hedgehog analysis helps, I thought about it in another sense which may be sort of useful at least as an analogy. 

A radius is a constant acceleration created by a constant force. So imagine if we are displacing a fluid whether it is air or snow or water , if the fluid is forced into having to move around a bend then going from zero to a constant acceleration instantly is jerky and inefficient and untidy. Why not have a controlled acceleration in a curve like a nicely controlled spline will do ??? I hazard to guess that this is relevant to snow and by forcing snow around corners in a brutal sort of instant acceleration results in less than ideal performance and feel......

Thanks again Mike for the help, I recon we are going in a good direction here even as a discussion its good to expand our knowledge and understanding and hopefully it leads to better board design for us long term, what ever we decide "better" is after these sort of discussions ...... It would be good to canvas some more wider input into this with others direct experience on this sort of thing whether it is wakeboard, windsorf, kiteboard, ski, snowboard etc etc as its all sort of relevant eh ?????